SintefManufacturing/python-urx #5: RobotException: Robot stopped

bauerj, 2016-04-27:
Whenever I execute a blocking command (e.g. `Robot.translate`), the library throws a `RobotException`, claiming the robot stopped.

The command, however, gets executed just fine (the robot moves to the right position).

```
>>> rob.translate((0.1, -0.1, 0), 0.7, 0.9)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\urx\urrobot.py", line 432, in translate
    return self.movex(command, p, vel=vel, acc=acc, wait=wait)
  File "C:\Python27\lib\site-packages\urx\robot.py", line 174, in movex
    return self.set_pose(t, acc, vel, wait=wait, command=command, threshold=threshold)
  File "C:\Python27\lib\site-packages\urx\robot.py", line 105, in set_pose
    pose = URRobot.movex(self, command, t.pose_vector, acc=acc, vel=vel, wait=wait, threshold=threshold)
  File "C:\Python27\lib\site-packages\urx\urrobot.py", line 322, in movex
    self._wait_for_move(tpose[:6], threshold=threshold)
  File "C:\Python27\lib\site-packages\urx\urrobot.py", line 211, in _wait_for_move
    raise RobotException("Robot stopped")
urx.urrobot.RobotException: Robot stopped
```

bauerj, 2016-04-27:
Okay, I found the problem. This library seems to be incompatible with Polyscope 3.2 at the moment.

I'll send a pull request with the necessary adjustments, if you don't mind.

oroulet, 2016-04-27:
That would be great. The parsing code is not very nice but should be very
easy to extend. How new is your robot? I have  a 6 months old robot and
everything works

bauerj, 2016-04-27:
Well, the robot I use is from September 2015, so it's even older than yours.

As it has got firmware from January 2016, it seems like someone did a firmware update.

oroulet, 2016-04-27:
OK. Maybe I need to update firmware too. Did you try to install math3d?
Installing it changes the algo for guessing move end. Much more robust

oroulet, 2016-04-27:
btw just print the robot object, it print the entire RobotModeData struct. values in this struct are used to know if robot is running. so this should be easy to fix

closed 2016-04-27 by bauerj

GraceKafuu, 2019-11-30:
rob.set_digital_out(1, False)
pose2 = (0.5006344557120612, 0.22362304931825686, 0.2581759195803068,0.8836140893706507, 2.9692676769852175,0.028800615397015658)
rob.movel(pose2, acc=a, vel=v, wait=True)

tried 11 times to find a packet in data, advertised packet size: -2, type: 3
Data length: 68
tried 11 times to find a packet in data, advertised packet size: -2, type: 3
Data length: 1092
tried 11 times to find a packet in data, advertised packet size: -2, type: 3
Data length: 1419
Traceback (most recent call last):
  File "/home/nvidia/UR/ur_1.py", line 22, in <module>
    rob.movel(pose2, acc=a, vel=v, wait=True)
  File "/home/nvidia/anaconda3/lib/python3.6/site-packages/urx/urrobot.py", line 284, in movel
    return self.movex("movel", tpose, acc=acc, vel=vel, wait=wait, relative=relative, threshold=threshold)
  File "/home/nvidia/anaconda3/lib/python3.6/site-packages/urx/robot.py", line 180, in movex
    return self.set_pose(t, acc, vel, wait=wait, command=command, threshold=threshold)
  File "/home/nvidia/anaconda3/lib/python3.6/site-packages/urx/robot.py", line 105, in set_pose
    pose = URRobot.movex(self, command, t.pose_vector, acc=acc, vel=vel, wait=wait, threshold=threshold)
  File "/home/nvidia/anaconda3/lib/python3.6/site-packages/urx/urrobot.py", line 316, in movex
    self._wait_for_move(tpose[:6], threshold=threshold)
  File "/home/nvidia/anaconda3/lib/python3.6/site-packages/urx/urrobot.py", line 217, in _wait_for_move
    raise RobotException("Robot stopped")
urx.urrobot.RobotException: Robot stopped

##########################################################
Why? Can anyone help me?

GraceKafuu, 2019-11-30:
Sloved!
##############################################
I had the same problem with UR5e (e-series) (UR Software 5.3.0.x). After using this fork of package:
https://github.com/jkur/python-urx/tree/SW3.5/urx

it works fine :)