SintefManufacturing/python-urx #8: get_digital_out doesn't work

bauerj, 2016-05-03:
I'm having problems finding out whether the gripper that is attached to the robot is open or closed.

I figured I could check `rob.get_digital_out(8, wait=True)` to find out if the digital out 8 (tool out 1) is on or off.

However, that doesn't seem to work:

### Gripper is open

[image]

```
>>> rob.get_digital_out(8, wait=True)
0
```

### Gripper is closed
[image]

```
>>> rob.get_digital_out(8, wait=True)
0
```

It returns 0 in both cases, although Polyscope shows that the output is on when the gripper is closed.

I checked that that's the correct output by using it to open/close the gripper with `python-urx` (which works great!).

oroulet, 2016-05-03:
I recently changed som stuff there. I see that in urxui I use
self.robot.get_digital_out_bits() and this one definitely work on the UR5 I
have here. will check this a newer ur10 next week.
Can you try get_digital_out_bits() and let me know?

bauerj, 2016-05-03:
No, that doesn't work either. It also returns 0 in both cases. 

I thought that the bitwise operations might be wrong but it seems the issue is somewhere else. I will check if it's the package parsing for `MasterBoardData`.

oroulet, 2016-05-03:
it works here, so it might be something with you newer firmware.

bauerj, 2016-05-03:
Yes, that's most likely the case here. It seems like the package parsing is perfectly fine, so maybe it's even a bug in the firmware of the robot?

oroulet, 2016-05-26:
Did you find out?

bauerj, 2016-05-27:
No, I did not. :confused: 

I'm using `ur.secmon.get_all_data()["ToolData"]["analogInput2"]` as a workaround as it seems to indicate the state of the tool too.

atomoclast, 2016-06-29:
I'm having a similar issue with the rob.get_analog_in(0, wait = False)

....we're getting wonky numbers. Is it the robot's firmware? Would rolling back solve this issue?

How do we use the ur.secmon.get_all_data() workaround exactly?

mtbrun, 2019-06-13:
I got the same problem with a 3.10 (23/05/2019) firmware on an UR10 CB3.
I don't think this is a robot firmware problem, the RealTime socket gives the right values for _actual_digital_input_bits_ and  _actual_digital_output_bits_.

The case for version >= 3.5 is not handled by the actual code. So all the datas in the  'MasterBoardData': {'size': 75,  'type': 3, ….. }  are binary shifted. 

In the ursecmon.py, it lacks the case for 3.5 and higher, to assign the good value to "self.version"

### Actual code (star line 66) :
```
 elif ptype == 0:
                # this parses RobotModeData for versions >=3.0 (i.e. 3.0)
                if psize == 38:
                    self.version = (3, 0)
                    allData['RobotModeData'] = self._get_data(pdata, "!IBQ???????BBdd", ("size", ...
                elif psize == 46:  # It's 46 bytes in 3.2
                    self.version = (3, 2)
                    allData['RobotModeData'] = self._get_data(pdata, "!IBQ???????BBdd", ("size", ....
                else:
                    allData["RobotModeData"] = self._get_data(pdata, "!iBQ???????Bd", ("size", ...
```
### Code modified
```
 elif ptype == 0:
                # this parses RobotModeData for versions >=3.0 (i.e. 3.0)
                if psize == 38:
                    self.version = (3, 0)
                    allData['RobotModeData'] = self._get_data(pdata, "!IBQ???????BBdd", ("size", ...
                elif psize == 46:  # It's 46 bytes in 3.2
                    self.version = (3, 2)
                    allData['RobotModeData'] = self._get_data(pdata, "!IBQ???????BBdd", ("size", ....
                elif psize == 47:  # It's 47 bytes in 3.5
                    self.version = (3, 5)
                    allData['RobotModeData'] = self._get_data(pdata, "!IBQ???????BBddi", ("size",
                else:
                    allData["RobotModeData"] = self._get_data(pdata, "!iBQ???????Bd", ("size", ...
```
With that it worked for me !