SintefManufacturing/python-urx #9: Compatible Software Versions with URX

atomoclast, 2016-06-29:
Hello all,

I am running URX to control a mix of UR10s and UR5s, and so far it has been easy and to interface with to control the robots. 

The movejs and movel commands work as expected and I can control the digital outs. I am now trying to read an analogIn from the robot, that I can clearly see on the pendant changing as expected from the sensor, but when I use rob.get_analog_in(0, wait = False)

....I am getting wonky numbers. Is it the robot's firmware? I saw in another issue someone mentioned it could be because the firmware someone was running was too new. Would rolling back solve this issue?

I checked the secmon.get_all_data()....it's showing the same wonky number that I get when I try and use get_analog_in...

If it is an issue with the with firmware version I am running, can you confirm which versions work for the UR10 and UR5?

Or maybe there's a work around? Other than this hiccup, this library is fantastic! 

Thanks!

oroulet, 2016-06-29:
In fact I wrote the method but I have never used it so I am not sure if it has ever worked...
@alvcap is supposed to upgrade our robots this summer so hopefully we will know soon if there is an issue there.
What number do you get? I get 0.0 when used on the fakerobot.py script which uses a dump from an old controller data.

alvcap, 2016-07-04:
Hi,
I have checked the method with the robots that we have:

- UR10, CB3, firmware 3.1.17336: get_analog_in **does not work**
- UR5, CB2, firmware 1.6.9671: get_analog_in **works**
- UR5, CB1, firmware 1.4.6899: get_analog_in **works**

In the case of the CB3, I get a very small exponential value, _wonky_ as @atomoclast mentioned.

alvcap, 2016-07-04:
Hi again,

From firmware 3.0 and onwards, UR has changed the data format for the robot state messages received using the secondary client interface.
https://s3-eu-west-1.amazonaws.com/ur-support-site/16496/Client_Interface.xlsx

I have updated my urx fork accordingly, and I am able to collect analog input data with the CB3 using firmware 3.1.17366. Check it if you are interested.

For previous versions of firmware, the code from the main urx repository works fine.

I guess that we will find a way to merge both solutions so that urx is compatible with all UR firmware.

oroulet, 2016-07-04:
@alvcap nice to see you are working :-) create a pull request so the changes get visible for everybody and I can can eventually merge them