SintefManufacturing/python-urx #11: Freedrive mode not working?

broofa, 2016-08-23:
Hi, I'm working with a UR5, trying to put it into freedrive mode.  URRobot.set_freedrive(True) does not seem to be working.  Reading the urx code it looks like it's sending `set robotmode freedrive`, however in reading the v3.2 of the URScript Programming Language guide, I don't see any mention of a `robotmode` command.  Am I missing something?

(Note: I'm new to this so my apologies if I'm overlooking something obvious.)

Cc: @markwsilliman

oroulet, 2016-08-28:
It used to work..., We'll check

zchen24, 2017-02-23:
In my lab, there are one UR3 running 3.1 and one UR5 running 1.8. This "set robotmode freedrive" command works on UR5, but does not work on UR3.

markwsilliman, 2017-02-23:
We're using this on a UR5 now that is running 3.4 so I would expect 3.1 to work.  One trick that we had to do is create a script to sleep after enabling freedrive.  Otherwise the program ends and freedrive turns off (all in < 1 second). 

Check your logs (on the robot) to see if the freemode is turning on and off.  If yes adding the sleep will fix it.

zchen24, 2017-02-23:
@markwsilliman Could you elaborate on "create a script to sleep after enabling freedrive"? 
On UR3, I also tried to call ```freedrive_mode()``` and found that the UR turns freedrive mode on, then turns it off after 16 ms.

markwsilliman, 2017-02-23:
Try something like:

 with urx.Robot(robot_ip) as rob:
            rob.send_program(b"def myProg():\n\tfreedrive_mode()\nsleep({})\nend".format(timeout))

zchen24, 2017-02-23:
WOW, works like a charm. Thank you!

PR #27 (2017-07-11): Change how freedrive is instantiated. Add freedrive timeout with default
Fixes #11 

@oroulet - This has been a long time coming so apologies.  Here is how we do freedrive at Tend.  The problem we faced is that the program ends as soon as it begins.  By adding a timeout (here it is 60 seconds but we use 3600 seconds internally) the program will remain running.  Hopefully others will find this useful.
closed 2017-07-11 by broofa