SintefManufacturing/python-urx #18: Timeout issues

MarcSallent, 2017-01-16:
Hello,

Just a note that I was having timeout issues and lost connections with my UR5 (not sure if the same one that you point in your 55ccfe7 commit).

I have been using @martinbjerge 's branch and have not had those issues yet. I have not done an extensive testing, but they should have appeared already.

I don't know if those issues are widespread or only I had them. If they are common, maybe you can get a solution from his fork.

Very useful package! Thanks!

AndyZe, 2017-08-15:
^This fork also fixed my timeout issue. Thanks!

AndyZe, 2017-08-17:
Actually, I'm finding out that the binary version doesn't have any connection issues at all. That's probably the easiest fix for others who are having this issue:

sudo apt install python-pip
sudo pip install urx