SintefManufacturing/python-urx #30: Compatibility with UR3 and polyscope 3.3.3.292

Wagdi-Benyaala, 2017-12-27:
Hello,

First, let me thank you for this python UR wrapper it is honestly a blessing given how easy it is to program in Python !

I recently got a UR3 arm and before asking I tested this wrapper with my arm. The get_robot.py example seems to work and when I print the robot object I get this : 
`Robot object is available as robot or r
Robot Object (IP=192.168.0.149, state={'isSecurityStopped': False, 'speedScaling': 1.0, 'isProgramPaused': True, 'timestamp': 2766576000, 'isRealRobotEnabled': True, 'isRobotConnected': True, 'robotMode': 7, 'isProgramRunning': False, 'isPowerOnRobot': True, 'isEmergencyStopped': False, 'speedFraction': 1.0, 'type': 0, 'controlMode': 0, 'size': 46})
INFO:urx:Closing sockets to robot
`
As specified in the title of this "issue" I am using a UR3 arm with Polyscope 3.3.3.292. Testing other examples like test_all.py moves the arm but I get a `urx.urrobot.RobotException: Robot stopped` on the python script and on the UR itself I get a `C153A2: Protective stop : position deviates from path : Elbow`.

The question here is : is this wrapper compatible with an UR3 or with Polyscope 3.3.3.292 (release October 2016). If yes, is there something I am doing wrong ? I also tried the urxui no problem to connect but always failing to control.

I looked a bit for this question before asking, if the answer is already somewhere here I really apologize for that. 

Thanks.

MzXuan, 2018-01-27:
Hello,
I am also testing this wrapper on UR3, with Polyscope 3.3.4.310 (which is newer than 3.3.3.292). I am a newbie to UR3 and python, but it seems this wrapper works with UR3.  I have tested `movel`, `movels`, `movej`, `speedl` and `stopj`. The only problem I have found is in urrobot.py, speedx function, 
`prog = "{}([{},{},{},{},{},{}], a={}, t_min={})".format(command, *vels)`
should be changed to
` prog = "{}([{},{},{},{},{},{}], a={}, t={})".format(command, *vels)`

I guess your C153A2 Protective stop is caused by invalid values given to the robot. It may help if you could tell which line/function cause the problem. Also the official explanation of the ur scripts is [here](https://s3-eu-west-1.amazonaws.com/ur-support-site/22198/scriptManual.pdf), hope this helps.

Sebbenbear, 2018-05-22:
Hi, I am running 3.5.2.10778, and I'm also getting the "urx.urrobot.RobotException: Robot stopped." The robot is definitely on and moves, but only for a short time before getting this error - is this a network issue?