SintefManufacturing/python-urx #31: does urx give joint speed?

MzXuan, 2018-01-30:
Hi,
I'm now using this driver to control UR3, and I also need joint speed data. The joint position data can get from`get_joint_data` function in `ursecmon.py`, but seems there is no method to get the joint velocity. Could anyone help with this?

Thanks

oroulet, 2018-01-31:
call get_all_data(). If data is not there then the robot probably does not expose it...

Mandelbr0t, 2018-12-15:
Hoi, you can get the joint speed out of rtmon if you parse more of the return string. For example:
`self._qActual = np.array(unp[31:37])
  self._qTarget = np.array(unp[1:7])
  self._tcp_force = np.array(unp[67:73])
  self._tcp = np.array(unp[73:79])
  self._tcp_speed = np.array(unp[61:67])`

for tcp speed. Have a look here:
[](http://www.zacobria.com/universal-robots-knowledge-base-tech-support-forum-hints-tips/knowledge-base/client-interfaces-cartesian-matlab-data/)