SintefManufacturing/python-urx #34: Fail to get gripper status

xiao-dong, 2018-03-13:
When I use RobotiqScript._get_gripper_status(), there is an error "TypeError: _socket_send_string() takes exactly 3 arguments (2 given)". 

I have no idea what is the message I should give in the function _socket_send_string(self, message, socket_name), which is inside the function_rq_get_var(self, var_name, nbytes).

Please help me.

DavidYaonanZhu, 2019-01-18:
When I want to control the robotiq gripper using the sample code given in readme, I get TypeError: __init__() takes at least 2 arguments (1 given). Please help me.

lixin-tj, 2021-04-21:
Hello, have you solved this problem?tks

lixin-tj, 2021-04-21:
> I have the same problem. Any updates?

Hello, have you solved this problem?tks

Xunmenggod, 2024-10-03:
hi, is there any update?

xiao-dong, 2024-10-03:
您好，您的邮件已接收。我稍后将与您联系，谢谢！吴华栋Wu Huadong中山大学 信息科学与技术学院