SintefManufacturing/python-urx #37: Remote control to send long code line by line using a python library

francescobic, 2018-05-25:
hi everyone.

i have an UR10 CB1.8 / 2.1 and i need to send long codes line by line.
i want to use this library written by @oroulet (https://github.com/oroulet).

does anyone know how to install it and where?

thanks a lot
Francesco

oroulet, 2018-05-31:
Another option for large programs is to keep the entire program in python but from python you get a stop between every movel commands so you will need to control your robot using velocity. This requires path planning, etc...

francescobic, 2018-05-31:
HI Olivier
thanks for your response and for your help

I want to send a code similar to those for 3D printing, it contains many variables on the path and must have continuity from beginning to end.
the file is in .script format and I thought that the best thing is to send the code a few lines at a time and to buffer with the memory of the robot while sending others.
through port 30003 I can also receive information on position and joint rotation of the robot, and do "hand shaking".
I think it's what your application does, but your application contains functions that generate the coordinates (if I'm not mistaken), but I would like to let him read my .script file, line by line.

I can send you my .script file, or upload like sample if you want

many thanks

oroulet, 2018-05-31:
you can always attach a sample so we understand. As far as I remember the Robot class has a send method you can use to send an UR script if this is what you want.

francescobic, 2018-05-31:
ok, here the code. 
sorry, what do you mean with robot class?
thanks.
[MainModule.script.zip](https://github.com/SintefRaufossManufacturing/python-urx/files/2057761/MainModule.script.zip)

oroulet, 2018-05-31:
looked at your files. you are probably completely overloading the robot controller with that kind of programs. You need to find a way to split it or implement you own motion controller. Looks like you need to learn some more programming then have a look at it again

francescobic, 2018-05-31:
ok thanks