SintefManufacturing/python-urx #41: movel and translate work on UR software 3.4 but on 3.5 yield "urx.urrobot.RobotException: Robot stopped"

raequin, 2018-07-31:
simple.py works fine on a UR10 that is running UR Software 3.4.4 but the Python script exits after the first motion command on a UR10 that is running 3.5.3, showing the following:

`Traceback (most recent call last):
  File "./simple.py", line 22, in <module>
    rob.movel(pose, acc=a, vel=v)
  File "/usr/local/lib/python2.7/dist-packages/urx/urrobot.py", line 284, in movel
    return self.movex("movel", tpose, acc=acc, vel=vel, wait=wait, relative=relative, threshold=threshold)
  File "/usr/local/lib/python2.7/dist-packages/urx/robot.py", line 180, in movex
    return self.set_pose(t, acc, vel, wait=wait, command=command, threshold=threshold)
  File "/usr/local/lib/python2.7/dist-packages/urx/robot.py", line 105, in set_pose
    pose = URRobot.movex(self, command, t.pose_vector, acc=acc, vel=vel, wait=wait, threshold=threshold)
  File "/usr/local/lib/python2.7/dist-packages/urx/urrobot.py", line 316, in movex
    self._wait_for_move(tpose[:6], threshold=threshold)
  File "/usr/local/lib/python2.7/dist-packages/urx/urrobot.py", line 217, in _wait_for_move
    raise RobotException("Robot stopped")
urx.urrobot.RobotException: Robot stopped
`

Mofeywalker, 2018-08-10:
I do experience the same problem with our URs on Firmware 3.5 and 3.6. The problem seems to be that, in ursecmon the variable "running" never gets True and the method _wait_for_move() in urrobot raises the exception in result of running being false

oroulet, 2018-08-10:
@alvcap Maybe you could update robot controller and test? We should also think if there is any way to make our code future proof. This is happening quite often...