SintefManufacturing/python-urx #48: UR5e communication issues

AlexKlimaj, 2018-09-05:
Hello,

I am trying to get this working with our new UR5e running URSoftware 5.0.5.30612

Any ideas? I will keep digging into it too. 

```
teal_robot_1@teal-robot-1-laptop:~/python-urx/examples$ python3 simple.py
WARNING:ursecmon:tried 11 times to find a packet in data, advertised packet size: -2, type: 3
WARNING:ursecmon:Data length: 68
WARNING:ursecmon:tried 11 times to find a packet in data, advertised packet size: -2, type: 3
WARNING:ursecmon:Data length: 1092
WARNING:ursecmon:tried 11 times to find a packet in data, advertised packet size: -2, type: 3
WARNING:ursecmon:Data length: 1444
robot tcp is at:  [-0.1920632364955505, 0.015287737270173504, 0.9737677686073115, -2.0645150325986665, 0.38708977551397505, 1.2446845282632593]
absolute move in base coordinate
Traceback (most recent call last):
  File "simple.py", line 20, in <module>
    rob.movel(pose, acc=a, vel=v)
  File "/home/teal_robot_1/.local/lib/python3.6/site-packages/urx/urrobot.py", line 284, in movel
    return self.movex("movel", tpose, acc=acc, vel=vel, wait=wait, relative=relative, threshold=threshold)
  File "/home/teal_robot_1/.local/lib/python3.6/site-packages/urx/robot.py", line 180, in movex
    return self.set_pose(t, acc, vel, wait=wait, command=command, threshold=threshold)
  File "/home/teal_robot_1/.local/lib/python3.6/site-packages/urx/robot.py", line 105, in set_pose
    pose = URRobot.movex(self, command, t.pose_vector, acc=acc, vel=vel, wait=wait, threshold=threshold)
  File "/home/teal_robot_1/.local/lib/python3.6/site-packages/urx/urrobot.py", line 316, in movex
    self._wait_for_move(tpose[:6], threshold=threshold)
  File "/home/teal_robot_1/.local/lib/python3.6/site-packages/urx/urrobot.py", line 217, in _wait_for_move
    raise RobotException("Robot stopped")
urx.urrobot.RobotException: Robot stopped
teal_robot_1@teal-robot-1-laptop:~/python-urx/examples$
```

AlexKlimaj, 2018-09-06:
I've got it sort of working. It will move but still throws errors. 

```
teal_robot_1@teal-robot-1-laptop:~/python-urx/examples$ python test_all.py
WARNING:ursecmon:tried 11 times to find a packet in data, advertised packet size: -2, type: 3
WARNING:ursecmon:Data length: 68
WARNING:ursecmon:tried 11 times to find a packet in data, advertised packet size: -2, type: 3
WARNING:ursecmon:Data length: 1092
WARNING:ursecmon:tried 11 times to find a packet in data, advertised packet size: -2, type: 3
WARNING:ursecmon:Data length: 1444
INFO:ursecmon:Remove 79 bytes of garbage at begining of packet
INFO:urx:Sending program: set_tcp(p[0, 0, 0, 0, 0, 0])
INFO:urx:Sending program: set_payload(0.5, (0,0,0))
('Digital out 0 and 1 are: ', 0, 0)
('Analog inputs are: ', (5.4588103927e-312, 5.284846273e-315))
('Initial joint configuration is ', [-0.6560118834124964, -1.7292429409422816, -1.067978858947754, -2.4921089611449183, 3.2261769771575928, -1.0817034880267542])
('Transformation from base to tcp is: ', <Transform:
<Orientation:
array([[ 0.40689847, -0.64679364,  0.64505165],
       [-0.26574904,  0.59179245,  0.76102506],
       [-0.87396286, -0.48108178,  0.06891465]])>
<Vector: (0.39471, -0.34685, 0.66703)>
>)
Translating in x
Click enter to continue

INFO:urx:Sending program: movel(p[0.444722,-0.346862,0.667029,-0.955128,1.168005,0.292974], a=0.3, v=0.05, r=0)
INFO:urx:Closing sockets to robot
Traceback (most recent call last):
  File "test_all.py", line 45, in <module>
    rob.translate((l, 0, 0), acc=a, vel=v)
  File "/home/teal_robot_1/.local/lib/python2.7/site-packages/urx/urrobot.py", line 429, in translate
    return self.movex(command, p, vel=vel, acc=acc, wait=wait)
  File "/home/teal_robot_1/.local/lib/python2.7/site-packages/urx/robot.py", line 180, in movex
    return self.set_pose(t, acc, vel, wait=wait, command=command, threshold=threshold)
  File "/home/teal_robot_1/.local/lib/python2.7/site-packages/urx/robot.py", line 105, in set_pose
    pose = URRobot.movex(self, command, t.pose_vector, acc=acc, vel=vel, wait=wait, threshold=threshold)
  File "/home/teal_robot_1/.local/lib/python2.7/site-packages/urx/urrobot.py", line 316, in movex
    self._wait_for_move(tpose[:6], threshold=threshold)
  File "/home/teal_robot_1/.local/lib/python2.7/site-packages/urx/urrobot.py", line 217, in _wait_for_move
    raise RobotException("Robot stopped")
urx.urrobot.RobotException: Robot stopped
teal_robot_1@teal-robot-1-laptop:~/python-urx/examples$

```

LukeSkypewalker, 2018-09-06:
Hi Alex, 

It is because of RTDE protocol difference between 3.4 and 3.5 versions (year++ ago).

There is a fork of URX with appropriate changes
https://github.com/jkur/python-urx/tree/SW3.5/urx

It should work fine on 5.x

Have fun

AlexKlimaj, 2018-09-06:
Thanks! That got me going.

oroulet, 2018-09-06:
Reopening we need to fix or merge

oroulet, 2018-09-06:
@jkur what about making a PR from your work? I can also make you a member of group or we can make a new group just for ur stuff

jkur, 2018-09-06:
Yes, merging would be great. I will have a look what needs to be done for a pull request. My fixes work but are not really reliable. I often get missed packets from the urx lib.

AlexKlimaj, 2018-09-06:
Yeah, I am getting this error on occasion.

```
teal_robot_1@teal-robot-1-laptop:~/Teal_Calibration_Robot$ python3 calibration_script.py
Traceback (most recent call last):
  File "calibration_script.py", line 9, in <module>
    rob = urx.Robot(ip_address)
  File "/usr/local/lib/python3.6/dist-packages/urx-0.11.0-py3.6.egg/urx/robot.py", line 27, in __init__
    URRobot.__init__(self, host, use_rt, use_simulation)
  File "/usr/local/lib/python3.6/dist-packages/urx-0.11.0-py3.6.egg/urx/urrobot.py", line 38, in __init__
    self.secmon = ursecmon.SecondaryMonitor(self.host, use_simulation)  # data from robot at 10Hz
  File "/usr/local/lib/python3.6/dist-packages/urx-0.11.0-py3.6.egg/urx/ursecmon.py", line 257, in __init__
    self.wait()  # make sure we got some data before someone calls us
  File "/usr/local/lib/python3.6/dist-packages/urx-0.11.0-py3.6.egg/urx/ursecmon.py", line 349, in wait
    raise TimeoutException("Did not receive a valid data packet from robot in {}".format(timeout))
urx.ursecmon.TimeoutException: Did not receive a valid data packet from robot in 0.5

```

oroulet, 2018-09-07:
Ok. Make a PR then we can do a review @alvcap

jkur, 2018-09-09:
Okay, sorry, i'm too slow. The fix is already in master.

closed 2018-09-10 by alvcap

raequin, 2019-03-28:
If somebody could tell me how to resolve this problem on 3.5 then I would really appreciate that.  The command `python3 urx_example.py` (which is almost nothing but `rob = urx.Robot("192.168.1.106")` yields the warning messages posted by @AlexKlimaj.