SintefManufacturing/python-urx #49: how to make the ur5 not slow down when it reach a point?

vannem95, 2018-09-18:
I need to move it in a trajectory and have it not slow down every-time it reached a point. Is there a way to set a final velocity at a point?
Thanks

vannem95, 2018-09-18:
@LukeSkypewalker  Thanks so much, man. Used movexs and it works well.
Do you happen to know how to slow it down at the final point? It stops quite abruptly at the final point with movexs.

LukeSkypewalker, 2018-09-18:
I do not remember details of movexs,
I used my own implementation.

If The main idea: middle points have to have a blend radius and the last one's radius should be zero.

But the last UR firmware (3.7) looks like fixes everything. Even wrong movep usage is smooth.
Make Upgrade!

Maybe I will find a code snippet for you later

vannem95, 2018-09-18:
Ok. Thanks