SintefManufacturing/python-urx #53: Sometimes UR3 gives timeout

pratikdeodhar, 2018-12-01:
[image]
This sometimes happens and sometimes not

axi92, 2019-08-27:
I have the same issue over and over...
I am on Version: 0.11.0
```
26-Aug-2019 20:17:23 | Exception in thread Thread-387:
-- | --
26-Aug-2019 20:17:23 | Traceback (most recent call last):
26-Aug-2019 20:17:23 | File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
26-Aug-2019 20:17:23 | self.run()
26-Aug-2019 20:17:23 | File "/home/user/.local/lib/python2.7/site-packages/urx/ursecmon.py", line 286, in run
26-Aug-2019 20:17:23 | data = self._get_data()
26-Aug-2019 20:17:23 | File "/home/user/.local/lib/python2.7/site-packages/urx/ursecmon.py", line 333, in _get_data
26-Aug-2019 20:17:23 | tmp = self._s_secondary.recv(1024)
26-Aug-2019 20:17:23 | timeout: timed out
```

axi92, 2019-08-30:
@pratikdeodhar did you found a solution? We have the error every night...

axi92, 2019-09-03:
I also tried the master and other forks but after some hours this Exeption stops everything all the time.
Is there any way to debug or get logs what is happening?

axi92, 2019-10-28:
We increased the timeout: https://github.com/axi92/python-urx/commit/1f375446fdd784f7381037318158aa7fdc96d5c4
It is maybe too high but it works now =)