SintefManufacturing/python-urx #54: Gripper is fully opened before going to target position.

masmura, 2018-12-03:
Environment:
UR5 + Robotiq 2-Finger Gripper Version 85
(Polyscope 3.7)

I tried to execute bellow script.
```
import sys
import urx
from urx.robotiq_two_finger_gripper import Robotiq_Two_Finger_Gripper

rob = urx.Robot("192.168.0.100")
robotiqgrip = Robotiq_Two_Finger_Gripper(rob)

robotiqgrip.close_gripper()
rob.close()
```
The robot's finger once opened, and closed after that.
My ideal action is to close its finger from the current position.

How can I make a robot work like that?
Thank you.

dominotasi, 2018-12-20:
Onrobotgripper has same issue, too, I solved with sleep. You can use time.sleep(2) after close_gripper function like this:

robotiqgrip.close_gripper()
time.sleep(2)
rob.close()

masmura, 2018-12-21:
Thank you for your reply.
I tried your advice but it didn't work for me...

kaixin-bai, 2020-02-26:
i also have that problem, and i found that urx has already done as what masmura said as the code below, but this won't fix the problem for me.
```
    def gripper_action(self, value):
        """
        Activate the gripper to a given value from 0 to 255

        0 is open
        255 is closed
        """
        urscript = self._get_new_urscript()

        # Move to the position
        sleep = 2.0
        urscript._set_gripper_position(value)
        urscript._sleep(sleep)

        # Send the script
        self.robot.send_program(urscript())

        # sleep the code the same amount as the urscript to ensure that
        # the action completes
        time.sleep(sleep)

    def open_gripper(self):
        self.gripper_action(0)

    def close_gripper(self):
        self.gripper_action(255)
```

lixin-tj, 2021-04-20:
Delete the follow statements in robotiq_ two_ finger_ gripper.py. 
        # Initialize the gripper
        urscript._set_robot_activate()
        urscript._set_gripper_activate()