SintefManufacturing/python-urx #57: No handlers could be found for logger "ursecmon"

WwYyFan, 2019-01-03:
Hi, I'm new to UR,Below is my code.sometime it works,but sometime it doesn't work.and terminal always output "No handlers could be found for logger "ursecmon" ",how to fix it?
code:
```
import sys
import urx
from urx.robotiq_two_finger_gripper import Robotiq_Two_Finger_Gripper
import time
if __name__ == '__main__':
    rob = urx.Robot("192.168.0.5")
    time.sleep(0.2)
    try:
        l = 0.1
        v = 0.07
        a = 0.1
        r = 0.05
        print rob.getl()
        rob.movel((0.01,0,0,0,0,0),a,v,relative=True)
        print rob.getl()
    finally:
	    rob.close()
	    print "true"
	    sys.exit()
```

oroulet, 2019-01-03:
That warning is not a bug, just some bad default configuration in very old python versions

WwYyFan, 2019-01-03:
@oroulet  thanks for reply, and I want to konw what ('Other: ', 0.9999559698671492) means?

DavidYaonanZhu, 2019-01-18:
I come up with the same issue, what does it mean? I use python 2.7.

oroulet, 2019-01-18:
https://stackoverflow.com/questions/44188270/no-handlers-could-be-found-for-logger

DavidYaonanZhu, 2019-01-18:
Then what does  ('Other: ', 0.9999559698671492) means?

brianFruit, 2019-04-30:
I'm having the same issue with the UR5. Looks like this ursecmon is used to monitor the UR at certain frequency. I think it is a symptom of a bigger issue. The "wait" in the movex functions are not working properly as the is_running status output by this ursecmon is always false. I still haven't found the issue yet, I'm hoping it's not hardware related.

kaixin-bai, 2020-01-08:
> I'm having the same issue with the UR5. Looks like this ursecmon is used to monitor the UR at certain frequency. I think it is a symptom of a bigger issue. The "wait" in the movex functions are not [...]

I also have that problem, did you fix it?