SintefManufacturing/python-urx #61: urx.urrobot.RobotException: Robot stopped

MarcSallent, 2019-02-26:
Hello,

This simple program always throws an

 ```
    test = RobotClass()
    logging.basicConfig(level=logging.INFO)
    dacc = 4
    dvel = 4
    ts = test.start("10.0.0.207")
    sleep(3)
    test.rob.movej((radians(90), radians(-70), radians(115), radians(-45), radians(90), radians(180)), acc=1, vel=1)
    sleep(1)
    test.stop()

```

liamstal, 2019-02-28:
I just learning now, but after moving the starting position of a real UR5 it did solve this problem.

bkinman, 2019-03-07:
What do you mean by it working after moving the starting position. Can you elaborate? I'm running into this same issue.

pxlong, 2019-06-05:
same issue here!  Could you please elaborate it? @liamstal

liamstal, 2019-06-05:
-Ubuntu 16.04-
I tested on real UR5.
generally, the problem for me was related to to the starting position of the TCP. it was relatively far/hard orientation reach from the first program starting position. 
I don't know why getting the robot to a somewhat more logical orientation and position allowed this error not to be thrown again.  maybe it was near a singularity, maybe its somehow related to position tolerances which I can't explain. 
again, just learning here, and it was some time ago.