SintefManufacturing/python-urx #64: urx.urrobot.RobotException: Robot stopped

Yasminbr, 2019-03-25:
hey, 
I try to run the following code: 
`from math import pi

import urx
import logging

if __name__ == "__main__":
    rob = urx.Robot("192.168.1.113")
    #rob = urx.Robot("localhost")
    rob.set_tcp((0,0,0,0,0,0))
    rob.set_payload(0.5, (0,0,0))
    try:
        l = 0.05
        v = 0.05
        a = 0.3
        j = rob.getj()
        print("Initial joint configuration is ", j)
        t = rob.get_pose()
        print("Transformation from base to tcp is: ", t)
        print("Translating in x")
        rob.translate((l, 0, 0), acc=a, vel=v)
        pose = rob.getl()
        print("robot tcp is at: ", pose)
        print("moving in z")
        pose[2] += l
        rob.movel(pose, acc=a, vel=v)

        print("Translate in -x and rotate")
        t.orient.rotate_zb(pi/4)
        t.pos[0] -= l
        rob.set_pose(t, vel=v, acc=a)
        print("Sending robot back to original position")
        rob.movej(j, acc=0.8, vel=0.2)

    finally:
        rob.close()`

it moves one timethe robot but than i receive the following errors: 
`/home/arl_ubuntu/PycharmProjects/Yasmin_new/venv/bin/python /snap/pycharm-community/117/helpers/pydev/pydevconsole.py --mode=client --port=37483
import sys; print('Python %s on %s' % (sys.version, sys.platform))
sys.path.extend(['/home/arl_ubuntu/PycharmProjects/Yasmin_new'])
Python 3.5.2 (default, Nov 12 2018, 13:43:14) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.2.0 -- An enhanced Interactive Python. Type '?' for help.
PyDev console: using IPython 7.2.0
Python 3.5.2 (default, Nov 12 2018, 13:43:14) 
[GCC 5.4.0 20160609] on linux
runfile('/home/arl_ubuntu/.PyCharmCE2018.3/config/scratches/scratch_7.py', wdir='/home/arl_ubuntu/.PyCharmCE2018.3/config/scratches')
tried 11 times to find a packet in data, advertised packet size: -2, type: 3
Data length: 44
tried 11 times to find a packet in data, advertised packet size: -2, type: 3
Data length: 68
tried 11 times to find a packet in data, advertised packet size: -2, type: 3
Data length: 1092
tried 11 times to find a packet in data, advertised packet size: -2, type: 3
Data length: 1419
Initial joint configuration is  [0.08181630820035934, -2.2004249731646937, 0.20488739013671875, 1.23224937915802, 0.5624825954437256, -3.4956069628344935]
Transformation from base to tcp is:  <Transform:
<Orientation: 
array([[-0.36767587, -0.87079988, -0.32637711],
       [ 0.47952922,  0.12317148, -0.8688386 ],
       [ 0.7967849 , -0.47595835,  0.37228681]])>
<Vector: (0.32584, -0.15352, 0.75274)>
>
Translating in x
Traceback (most recent call last):
  File "/home/arl_ubuntu/PycharmProjects/Yasmin_new/venv/lib/python3.5/site-packages/IPython/core/interactiveshell.py", line 3267, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-2-199644d82eb9>", line 1, in <module>
    runfile('/home/arl_ubuntu/.PyCharmCE2018.3/config/scratches/scratch_7.py', wdir='/home/arl_ubuntu/.PyCharmCE2018.3/config/scratches')
  File "/snap/pycharm-community/117/helpers/pydev/_pydev_bundle/pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "/snap/pycharm-community/117/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/home/arl_ubuntu/.PyCharmCE2018.3/config/scratches/scratch_7.py", line 20, in <module>
    rob.translate((l, 0, 0), acc=a, vel=v)
  File "/home/arl_ubuntu/PycharmProjects/Yasmin_new/venv/lib/python3.5/site-packages/urx/urrobot.py", line 429, in translate
    return self.movex(command, p, vel=vel, acc=acc, wait=wait)
  File "/home/arl_ubuntu/PycharmProjects/Yasmin_new/venv/lib/python3.5/site-packages/urx/robot.py", line 180, in movex
    return self.set_pose(t, acc, vel, wait=wait, command=command, threshold=threshold)
  File "/home/arl_ubuntu/PycharmProjects/Yasmin_new/venv/lib/python3.5/site-packages/urx/robot.py", line 105, in set_pose
    pose = URRobot.movex(self, command, t.pose_vector, acc=acc, vel=vel, wait=wait, threshold=threshold)
  File "/home/arl_ubuntu/PycharmProjects/Yasmin_new/venv/lib/python3.5/site-packages/urx/urrobot.py", line 316, in movex
    self._wait_for_move(tpose[:6], threshold=threshold)
  File "/home/arl_ubuntu/PycharmProjects/Yasmin_new/venv/lib/python3.5/site-packages/urx/urrobot.py", line 217, in _wait_for_move
    raise RobotException("Robot stopped")
urx.urrobot.RobotException: Robot stopped
`

TomaszMolik, 2019-06-06:
I had the same problem with UR5e (e-series) (UR Software 5.3.0.x). After using this fork of package:
https://github.com/jkur/python-urx/tree/SW3.5/urx

it works fine :)

nouyang, 2019-07-10:
Hi, I also had this problem, and realized I was running the `pip install urx` version, which installs an old version of this package that does not have the changes for the 3.5 UR firmware. Instead, I import urx directly from this repository and it works (I think this repo has merged many of the changes in jkur fork already?).

You can check the `ursecmon.py` file to make sure it has something about psize == 47. e.g.

```
                elif psize == 46:  # It's 46 bytes in 3.2
                    self.version = (3, 2)
                elif psize == 47:
                    self.version = (3, 5)
```

GraceKafuu, 2019-11-30:
> I had the same problem with UR5e (e-series) (UR Software 5.3.0.x). After using this fork of package:
> https://github.com/jkur/python-urx/tree/SW3.5/urx
> 
> it works fine :)

Thank you ! 
It works fine!

MaeliseCastel, 2020-07-03:
Hi ! I'm new in all of it... Can someone tell me how I can take this new package and make it work as a casual python library please ?

nouyang, 2020-07-03:
@MaeliseCastel  Can you describe your issues in more detail? e.g. what have you tried

MaeliseCastel, 2020-07-03:
> @MaeliseCastel Can you describe your issues in more detail? e.g. what have you tried

I'm not currently at work but I can describe it and show you the entire error monday. 
I used the pip install urx instruction to have the package to controll my UR5e robot. It seems that some functions work such as get_pos() but I can't use the move functions because it says that the robot stopped or something like this.

Thank you for your fast answer ! Have a nice week end !

TomaszMolik, 2020-07-04:
Try this:
http://lab.iqrobots.eu/universal-robots/universal-robots-remote-control-using-python-urx-library/

MaeliseCastel, 2020-07-06:
> Try this:
> http://lab.iqrobots.eu/universal-robots/universal-robots-remote-control-using-python-urx-library/

Unfortunately it doesn't work, I have another error now. This is the following one : 
[image]

MaeliseCastel, 2020-07-06:
But I have this error only when i try to run it for the first time. When i do some changes and I rerun the programm, this error appears but when I run it again 1min later, it's gone... I don't know what it's about

But thank you all for your fast answers !

MaeliseCastel, 2020-07-07:
Is it possible that what worked yesterday doesn't work anymore ? Because I didn't change anything and this morning it doesn't work, nothing happens

kritzikratzi, 2020-07-07:
just keep pressing play. sometimes the connection doesn't come up quick enough. it seems there are some forks of python-urx that fix this problem, but just pressing play multiple times also solves it :)

nouyang, 2020-07-07:
Oh, yes, I just increased the time outs manually ^^; Sorry to see this just now. I will try to dig out my values later, maybe they are:
```
PACKET_TIMEOUT = 1.0
FIND_FIRST_PACKET_ATTEMPTS = 101
```

MaeliseCastel, 2020-07-08:
> just keep pressing play. sometimes the connection doesn't come up quick enough. it seems there are some forks of python-urx that fix this problem, but just pressing play multiple times also solves it :)

It doesn't work for me unfortunately

nouyang, 2020-07-17:
As per https://github.com/SintefManufacturing/python-urx/issues/84 I recommended sending URScript directly over serial, as in this gist: https://gist.github.com/nouyang/bac35bee6712855a4513eaaf53e5b923

TomaszMolik, 2020-07-18:
You can also consider using this library: https://sdurobotics.gitlab.io/ur_rtde/index.html

kritzikratzi, 2020-07-18:
@TomaszMolik ur_rtde is also very good indeed. 

@MaeliseCastel how are you connected to the robot? for me, anything other than a direct wired connection causes tons of problems. you could try another computer to be sure, maybe your system network settings are strange.

maryviktory, 2021-02-10:
> Oh, yes, I just increased the time outs manually ^^; Sorry to see this just now. I will try to dig out my values later, maybe they are:
> 
> ```
> PACKET_TIMEOUT = 1.0
> FIND_FIRST_PACKET_ATTEMPTS = 101
> ```

 @nouyang, Hi, where do you have these values, are those in URX library or you code them in, please clarify how to implement

nouyang, 2021-02-17:
@maryviktory I actually don't remember now, probably these lines
https://github.com/jkur/python-urx/blob/SW3.5/urx/ursecmon.py#L206
https://github.com/jkur/python-urx/blob/SW3.5/urx/ursecmon.py#L341
Good luck, do let issue tracker know if that works.

oroulet, 2021-02-18:
I do not have access to these robots anymore but if someone has a branch that works. I can put him as administrator here and he can merge his work.

majunqiao, 2021-11-16:
> 但是只有当我第一次尝试运行它时才会出现这个错误。当我做了一些更改并重新运行程序时，会出现此错误，但是当我在 1 分钟后再次运行时，它消失了......我不知道它是关于什么的
> 
> 但感谢大家的快速答复！
Hello, I have encountered the same problem recently. How does urx control ur5e