SintefManufacturing/python-urx #65: Packet data problems

raequin, 2019-03-29:
First of all, please let me know if this is not the proper forum for my problem (and where I might go instead).  This is in regards to a UR 10 running 3.5.3.10825.  For the script urx_get_robot.py

```
import urx
import logging

if __name__ == "__main__":
    try:
        logging.basicConfig(level=logging.INFO)
        robot = urx.Robot("192.168.1.106")
        
    finally:
        robot.close()
```

When I run the command python3 urx_get_robot.py, here is what I get.

```
WARNING:ursecmon:tried 11 times to find a packet in data, advertised packet size: -2, type: 3
WARNING:ursecmon:Data length: 68
WARNING:ursecmon:tried 11 times to find a packet in data, advertised packet size: -2, type: 3
WARNING:ursecmon:Data length: 1092
WARNING:ursecmon:tried 11 times to find a packet in data, advertised packet size: -2, type: 3
WARNING:ursecmon:Data length: 1418
INFO:ursecmon:Remove 79 bytes of garbage at begining of packet
INFO:urx:Closing sockets to robot
```

It's unknown to me whether this is a problem with the robot version and urx or if the UR 10 is behaving abnormally, because when I run the following script

```
import socket

HOST = "192.168.1.106"    # The remote host
PORT = 30002              # The same port as used by the server
s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.connect((HOST, PORT))

for i in range(2):
    data = s.recv(32768)
    print ("Received", data)

s.close()
```

This is what prints:

```
Received b'\x00\x00\x007\x14\xff\xff\xff\xff\xff\xff\xff\xff\xfe\x03\tURControl\x03\x05\x00\x00\x00\x03\x00\x00\x00\x0015-02-2018, 13:17:59\x00\x00\x00\x18\x14\xff\xff\xff\xff\xff\xff\xff\xff\xfe\x0c\x00\x00w$\x00\x00w$\x01'
Received b"\x00\x00\x05F\x10\x00\x00\x00/\x00\x00\x00\x00\x00]\xdbX\x00\x01\x01\x01\x00\x00\x00\x00\x07\x00?\xf0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00?\xf0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfb\x01\xc0\x07\x89e\x88\x88Z0\xc0\x07\x89r\x08\x88Z0\x00\x00\x00\x00\x00\x00\x00\x00\xbeR\xf0\x01B@/\x1bA\xd7\xeb\x81A\xde\x9e'\xfd\xc0\x00xWh\x88Z0\xc0\x00xc\xe8\x88Z0\x00\x00\x00\x00\x00\x00\x00\x00@\xa2\x99\x01B@\x00\x00A\xe2\xcc\xc8A\xf6\xbd\xb3\xfd\xbf\xf9\xd2\xecQ\x10\xb4`\xbf\xf9\xd3\x11\xd1\x10\xb4`\x80\x00\x00\x00\x00\x00\x00\x00@K\x97\x9fBA\x85\x1fA\xea\x14vA\xfb\x14\xdd\xfd\xc0\x04\xff'\xe8\x88Z0\xc0\x04\xff'\xe8\x88Z0\x00\x00\x00\x00\x00\x00\x00\x00=\xd3P\x01B@]/A\xecz\xddB\x13\xf9I\xfd\xbf\xf43\xb6Q\x10\xb4`\xbf\xf43\xdb\xd1\x10\xb4`\x00\x00\x00\x00\x00\x00\x00\x00<\xee\xe0\x02B@|\xeeB\x04\n9B\x08\x12B\xfd\xbf\xf9[#\xd1\x10\xb4`\xbf\xf9[=Q\x10\xb4`\x00\x00\x00\x00\x00\x00\x00\x00>\x1e|\x01B@|\xeeB\x02O\xcfB\x14\xf3\xe1\xfd\x00\x00\x00e\x04\xbf\xecW9\xf6v8~?\x86E\xben\xda\x1d\xcf?\xd0\xb5\x01\x13\xf0\xdd\x9a\xc0\x01r\xf8\xc5\xdf\xe30?\xbb\xe0\x81\x1c\xd0=.@\x01)\x90m\ni\xdc\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xe1\x05l\xe9\xb4/\xac(\xcf\x8b\x8b\x06\xfewZ\xee\x8f\x80\xab\xa6\r\x05\x96\x94]w?+\xdetTHF)\xbf\xcf\x91\xe94i$\x17?\xd2\x95\xf0\xc7\x04\xe8f\xbf\xa6:\x9fBx7\x9b\xbf9\xb66k\xf0\x19\xd2\xbfG\x1c\x80\xb58>K?\x01E,o7#\x19\xbf\xe3\x00$r\x96\xbc\xb0\xbf\xe2A\x05\xca\xf2Ps?\x05\xcf\xe3\x81\x92^\xbc\xbe\xea\x12\x1cX\xca^\xe7\x00\x00\x00\x00\x00\x00\x00\x00?\xc0f\xabF\xba\xab,@68\xe3%\x06l\xbf\xc08\xf3\x0f\xa6\x14\x87\xdf@\x07\x1f\xde*\xa2b3?\xbd\x9a\xdb\xa4\\\xaez?\xb7\x98\x92\x06\xcd,\xc1?\xf9!\x81\tW\xbfd\xbf{\x94\xb1\xdb\xc5\x7fp\xbf\x82yTQk\x119?\xf9 \xa3\xb7-\xdd\xe7\xbf\xf9$\xafS\xac\xb9\x9a\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x00\x005\t@\nC\x04\xe3\xe50\x19\xbf\xf0ZE\x81R\xb3a\xc0\x0e\xf4'\xe1N\x07b\xbf\xe8\x01/\xef\x00v?\xbf\xd7\xfb\xe9:\x88\xf3\xd1\xc0\x00\x0fi\x90'}[\x00\x00\x00J\x03\x00\x00\x00\x00\x00\x00\x00\x04\x01\x01?p\x7f\xc0\x00\x00\x00\x00?p\x7f\xc0\x00\x00\x00\x00\x00\x00?pbM\xe0\x00\x00\x00?pbM\xe0\x00\x00\x00A\xfc \x00BA~>?\x80UD>\\\x82\xd3\x01\x00\x00\xc2D[d\x00\x00\x00\x00\x00%\x02\x01\x01?\xa7\x1b\xfd@\x00\x00\x00?\xa6\xa1\x96\xe0\x00\x00\x00C\xe5[2\x00\x00\x00\x00\x00B@\x00\x00\xfd\x00\x00\x01\xbd\x06\xc0\x19W\x99(,/c@\x19W\x99(,/c\xc0\x19W\x99(,/c@\x19W\x99(,/c\xc0\x19W\x99(,/c@\x19W\x99(,/c\xc0\x19W\x99(,/c@\x19W\x99(,/c\xc0\x19W\x99(,/c@\x19W\x99(,/c\xc0\x19W\x99(,/c@\x19W\x99(,/c@\x02Z\xeb\xd1\xc7\x0c\xff@D\x00\x00\x00\x00\x00\x00@\x02Z\xeb\xd1\xc7\x0c\xff@D\x00\x00\x00\x00\x00\x00@\n\xbb\x94\xed\xdd\xc6\xb1@D\x00\x00\x00\x00\x00\x00@\n\xbb\x94\xed\xdd\xc6\xb1@D\x00\x00\x00\x00\x00\x00@\n\xbb\x94\xed\xdd\xc6\xb1@D\x00\x00\x00\x00\x00\x00@\n\xbb\x94\xed\xdd\xc6\xb1@D\x00\x00\x00\x00\x00\x00?\xf0\xc1R8-se?\xf6W\x18J\xe7D\x87?\xd0\x00\x00\x00\x00\x00\x00?\xf3333333?\xd0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xbf\xe3\x95\x81\x06$\xdd/\xbf\xe2PH\x16\xf0\x06\x8e\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00?\xc0K]\xccc\xf1A\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00?\xc4\xfc\x04\xc8\xbc\x9c\xd4?\xbd\x9e\x83\xe4%\xae\xe6?\xb7\x9akP\xb0\xf2|?\xf9!\xfbTREP\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00?\xf9!\xfbTREP\xbf\xf9!\xfbTREP\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06\x00\x00\x00\x05\x00\x00\x00\x02\x00\x00\x00\x02\x00\x00\x00=\x07\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00?\xb1B\x92\x7fK\xe2\xea\x00\x00\x00\x08\x08\x00\x01\x00\x00\x00\x00+\n\xc2D[d\x00\x00\xbf\xd1\xff\xb4W\xae\x12[\xbf\xad\x1d5^\xed\xcae?\xe5_7\x1ff\x17\xd1\x00\x00\x00\x00\x00\x00\x00\x00"
```

No matter how many times the loop repeats there are always similar characters printed.  Should the robot be continually sending data like that?  Any help you can give would be so so appreciated.

DarthThomas, 2019-06-04:
Got same issue

nouyang, 2019-07-10:
I solved the issue by uninstalling the `pip install urx` version, and instead using this repo directly. See
https://github.com/SintefManufacturing/python-urx/issues/64#issuecomment-510213146