SintefManufacturing/python-urx #71: Retrive data values using UR script

shyam2718, 2019-05-15:
I am able to manipulate robot using movel, movej commands via urx.send_message
Example. 
robo = urx.Robot("IP of the robot")
robo.send_message("movej([0,-1.5707,1.5707,0,0,0],a = 0.05,v = 0.05)")
Likewise, i am interested in knowing the values of get_actual_tcp_pose(), get_inverse_kin() etc through urx.  Can someone help me on this

oroulet, 2019-05-15:
urx simply implement client side of URScript, so either URScript has these methods or they do not exists. but there is Robot.get_pose() and Robot.get_joints() (or similar I do not remember exactly the API). But there is no way to know the current TCP setup, this is missing in URScript API

shyam2718, 2019-05-15:
> urx simply implement client side of URScript, so either URScript has these methods or they do not exists. but there is Robot.get_pose() and Robot.get_joints() (or similar I do not remember exactly the API). But there is no way to know the current TCP setup, this is missing in URScript API

The functions that I have mentioned are part of URScript. So connecting the robot through socket communication is only way to retrieve the values? or is there any command in urx that would fetch the values.

oroulet, 2019-05-15:
as far as I remember you cannot get the output of a cript/Command thourgh TCP. but the robot streams a lot of information. so you can call Robot.secmon.get_all_data() and see if the data you want is there in that big dict

shyam2718, 2019-05-15:
> as far as I remember you cannot get the output of a cript/Command thourgh TCP. but the robot streams a lot of information. so you can call Robot.secmon.get_all_data() and see if the data you want is there in that big dict

Thanks. is there any document available for URx. There are commands in URx, which is not there in
URScript. ex. movex.

oroulet, 2019-05-15:
no. no docs apart from the docstrings. I added movex but I cannot remember why ;-) and I do not work with that robot anymore