SintefManufacturing/python-urx #73: Error when using speedj, "speedj() got an unexpected keyword argument 't_min'"

SunYuansong, 2019-07-24:
Hi,
I want to try the speedj function by the following code:

`import urx`
`import time
``if __name__ == '__main__':`
    `rob = urx.Robot('192.168.1.103')`
    `while True:`
        `  rob.speedj([0.2,0.3,0.1,0.05,0,0], 0.5, 0.5)`
        `  time.sleep(0.5)`
    `rob.close()`

However, the teach pad gives this error:
[image]

I have tried different t_min in the speedj but did not make any difference.
Is there any solution for this issue? Any suggestion would be grateful.

Abdul-UOiT, 2019-10-05:
@SunYuansong did you solve the issue ?

SunYuansong, 2019-10-07:
> @SunYuansong did you solve the issue ?
No, I can not find the problem, it seems the urx speedj command has some bugs with the new ur system version.
I am using ros_control now which can also realize real time speed control.

Abdul-UOiT, 2019-10-07:
@SunYuansong could you please explain how do u use ros_control 
I am trying to run the joystick [here](https://github.com/SintefManufacturing/python-urx/blob/master/examples/joystick_control.py) example with UR 3.3

SunYuansong, 2019-10-07:
It needs some ROS knowledge, I can explain the process roughly, but if you want to go into the details you may need to check the ros_control github page.
First, you need to start the joint speed controller node. If you have used moveit to control ur, you must have used the ur modern driver. The default .launch file will start a joint_trajectory_follow_controller (I'm not sure about the controller name, maybe similar), here you need to change it to the joint_speed_controller (also you need to check the real controller name). And after you start the controller, there will be a ros topic waiting for joint velocity data.
Second, in your joystick control program, you need to publish the joint velocity to the topic mentioned above. Here you need to pay attention to the format of the joint velocity data, it must fit the topic.

Abdul-UOiT, 2019-10-07:
Yeah by using `ur_modern_driver`, you can always send joint velocities on the joint trajectory topics. `jog_arm_server` also uses speedj. But this has nothing to do with this repo `python-urx`

BartMoyaers, 2019-11-19:
I'm a little late to the party, but: you can fix this error by changing the source code on the following line:
https://github.com/SintefManufacturing/python-urx/blob/96303c0de7df46ebc6dede562ee9f4b1c34f2504/urx/urrobot.py#L266

Just change it to not use the keyword arguments:
`prog = "{}([{},{},{},{},{},{}],{},{})".format(command, *vels) `

We had the same problem with our UR5 using the 3.5 firmware.