SintefManufacturing/python-urx #74: How to get a message to know if the robot has successfully moved to the defined position?

kaixin-bai, 2019-10-08:
just found this below in example code, is it possible to get a message to know whether the robot has successfully moved to the defined position, because the robot move sometimes 2s and sometimes 6s, but i don't want the robot always wait for a long time.
```
print("Waiting 2s for end move")
time.sleep(2)
```

Thanks a lot

alejandro-colomar, 2020-03-31:
The robot does send a message through tcp, which contains its position and speed.  It is internally managed by this library (I don't know exactly how because I'm not an expert in python, but more or less I understand the internals).  However, you shouldn't need it, as the movement functions already do that for you and only ask for a timeout:

```
    def movel(self, tpose, acc=0.01, vel=0.01, wait=True, relative=False, threshold=None):
        """
        Send a movel command to the robot. See URScript documentation.
        """
        return self.movex("movel", tpose, acc=acc, vel=vel, wait=wait, relative=relative, threshold=threshold)
```
Movement functions internally call this function, which in its comment explains the situation:

```
    def _wait_for_move(self, target, threshold=None, timeout=5, joints=False):
        """
        wait for a move to complete. Unfortunately there is no good way to know when a move has finished
        so for every received data from robot we compute a dist equivalent and when it is lower than
        'threshold' we return.
        if threshold is not reached within timeout, an exception is raised
        """
```

The UR sends its message through TCP at a fixed frequency of 10 Hz (BTW, that's the maximum frequency at which it can receive commands, or it will go nuts).

closed 2021-10-04 by kaixin-bai

alejandro-colomar, 2021-10-04:
Yes, the robot sends a status message every 0.1 seconds that's contains, among other fields, the position and velocity of the robot, so you can read that message every 0.1 seconds to have a bit more precise behavior

alejandro-colomar, 2021-10-04:
You can check my C library for UR, although I haven't had access to a UR for a year, so it is untested, and probably has important bugs, but may give you some idea of how it should work:

https://github.com/alejandro-colomar/libalx/tree/main/src/robot-ur