SintefManufacturing/python-urx #77: Goal not reached but no program has been running for 5 seconds.

kaixin-bai, 2019-10-27:
i'm using ur5 now, with 'movej' it works sometimes, but the error below occurs sometimes. I found that someone says change the settings to remote control to enable will help for ur-e serie, while i can't find that setting with ur5. Can anyone help me? Thanks! 
 `raise RobotException("Goal not reached but no program has been running for {} seconds. dist is {}, threshold is {}, target is {}, current pose is {}".format(timeout, dist, threshold, target, URRobot.getl(self)))
urx.urrobot.RobotException: Goal not reached but no program has been running for 5 seconds. dist is 1.08735961347, threshold is 0.870018245315, target is [ 3.03882384 -1.97228843  1.8254056  -1.37555343 -1.61774379  1.10942733], current pose is [0.2802325137473805, 0.15454954432829898, -0.10841777566869056, -0.22737147091710713, 3.1121640387628653, 0.021905580945764575]
`

kaixin-bai, 2020-01-08:
https://github.com/SintefManufacturing/python-urx/issues/62

closed 2020-01-08 by kaixin-bai

majunqiao, 2021-11-17:
hello，I have a same question，How did you solve it