SintefManufacturing/python-urx #84: Gripper not activated

MaeliseCastel, 2020-07-06:
Hi all, 

I'm working on a UR5e and I'm new in all of it. I've managed to move the robot and everything but now I try to implement something with a robotic gripper. However when I run my code (a very simple one as 
rob = urx.Robot("192.168.0.100")
gripper = Robotiq_Two_Finger_Gripper()
gripper.close_gripper() )
It doesn't work, i don't have any errors but the gripper doesn't close and there isn't the blue light saying that it is on. 
Did anyone have an idea of what I can do please ?

MiriamEichinger, 2020-07-06:
Hi maelise,
Unfortunately I am still using polyscope most of the time, but I also run into that error in polyscope. There you have to activate the gripper before u can either close or open it. That solution sounds way to easy... I know. Maybe u need to initialise(probably the wrong word :)) before u can us it.

MaeliseCastel, 2020-07-07:
Hi Miriam, 

Thanks for you answer. I tried to activate it manually before using the remote control but it still doesn't work. 
I will try to find a way to initialise it

kritzikratzi, 2020-07-07:
@MaeliseCastel  are you using exactly the "two finger gripper" by robotiq, or something else?

MaeliseCastel, 2020-07-07:
> @MaeliseCastel are you using exactly the "two finger gripper" by robotiq, or something else?

I am exactly using the Robotiq 2-Finger Adaptative Robot Gripper

kritzikratzi, 2020-07-07:
okay. trivial question, but is it possible you just forgot to pass the robot object to the hand? ie. 

    rob = urx.Robot("192.168.0.100")
    # note the "rob" object being passed here
    gripper = Robotiq_Two_Finger_Gripper(rob)
    gripper.close_gripper()

MaeliseCastel, 2020-07-08:
I did it already but thanks for trying :) 
I think that I will try to go on RoboDK for my project because it seems easier but thank you for your answers !

MaeliseCastel, 2020-07-09:
Well RoboDK is not what I was looking for so I'm coming back here ! If anyone has an idea of the problem, feel free to talk with me ! 
Could it be because of the IP and port that urx uses for the connection with the gripper ? 
"""The gripper is controlled by connecting to the robot's computer (TCP/IP) and
then communicating with the gripper via a socket (127.0.0.1:63352).  The scope
of the socket is at the program level.""" (cf. robotiq_two_fnger_gripper.py)

nouyang, 2020-07-09:
Hello,
I suffered a lot of this pain. You are a grad student correct? If it's helpful I can hop on a video call, and we can report back any results. Please directly contact me at nrobotspam at mit dot edu (with the spam removed). 

Of note there is also a ROS package which you may have more luck with than this pure python approach. It's also possible to directly send URscript over.

nouyang, 2020-07-17:
Hi, to circle around, we didn't hop on a call but I did recommend sending a urscript directly over serial, instead of using this python library, as it appeared to be a short timeframe project. I'm not sure if Maelise will report back on the final solution for this specific issue, but leaving here if of interest:
https://gist.github.com/nouyang/bac35bee6712855a4513eaaf53e5b923

MaeliseCastel, 2020-07-20:
I found a way to move the gripper with the urx library ! 
You'll have to change just one thing : in the file robotiq_two_fnger_gripper.py, change the IP address (127.0.0.1) with the IP address of your robot ! If I'm not clear, ask me again.

It works for me. I manage to close and open the gripper and even to have other positions. 

However for the rest of my programm, I had to change some things in this file but it's just for me, it was only after that I moved the gripper. So normally you just have to change the IP address by your robot's one.

Thank you all for trying your best to help me !

kritzikratzi, 2020-07-20:
@MaeliseCastel  sorry, but this seems very wrong. glad it works for you. but it's definitely not needed to make changes to the library in this way (and it defeats the whole point of having a library in the first place), but hey ... whatever works i guess

MaeliseCastel, 2020-07-20:
> @MaeliseCastel sorry, but this seems very wrong. it's definitely not needed to make changes to the library in this way (and it defeats the whole point of having a library in the first place), but hey ... whatever works i guess? :/

Hi, 

I'm still totally new in robotics so could you please tell me why is that wrong ? 
And why does it defeats the whole point of having a library in the first place please ?

I guess that I need to learn more about all of it :+1:

kritzikratzi, 2020-07-20:
you shouldn't have to modify the library. the point of having a library is to install it, and then use it. 
imagine there is an update to the library, you would always have to keep track of your changes and apply them again. 

the code i sent you works for me: 

```
rob = urx.Robot("192.168.0.100")
# note the "rob" object being passed here
gripper = Robotiq_Two_Finger_Gripper(rob)
gripper.close_gripper()

```

but if that doesn't work, you can also pass the hostname as a parameter to the gripper code in this way: 
```
robot_ip = "192.168.0.100"
payload=0.85
speed=255
force=50
rob = urx.Robot(robot_ip)
# note the "rob" object being passed here
gripper = Robotiq_Two_Finger_Gripper(rob, socket_host=robot_ip)
gripper.close_gripper()

```

you can find the gripper constructor and all it's parameters here: https://github.com/SintefManufacturing/python-urx/blob/master/urx/robotiq_two_finger_gripper.py#L142

MaeliseCastel, 2020-07-20:
I doesn't work for me and I don't exactly have a library because I have an e-series so I had to clone another branch of this git hub. But you're right that I should just put it in the constructor.

closed 2020-07-21 by MaeliseCastel

kefusong, 2022-12-10:
Hi, I had the same problem as you before, after using your method, it didn't work.
But when I modify the code in robotiq_two_fnger_gripper.py which is  urscript._set_tool_voltage(24), it works!
Oh, I must mention that I am using UR16e and the handE gripper, hope this can help others

Xunmenggod, 2024-09-30:
Hi, I just started to use robotiq gripper and urx package. I would like to know if there is a method to achieve the grip force control of the gripper since I will use it to grasp some fragile objects. I tried to send a urscript that aims to set the speed of force value for the gripper, but it still seems like position control.

AlejandroSilvestri, 2024-10-24:
Hi @Xunmenggod ,
You can set the force in the constructor:
https://github.com/SintefManufacturing/python-urx/blob/e277ef1a7fb057f9e642a21cd95ade2b4e5a3468/urx/robotiq_two_finger_gripper.py#L146

or change the force parameter after construction.

BTW, can you control de gripper from a PC?  I used to control it with this library, but suddenly it stop working.  The gripper works from the teach pendant, and I can control the robot by using URX, but not the gripper.