SintefManufacturing/python-urx #102: Does movej support URSim?

wly2014, 2021-11-03:
Does the `movej` support URSim? Although I can get the state right, why I always get this error when using `movej`:

> Current tool pose is:  [-0.11606565366751193, -0.05602054660042595, -0.01930032084546665, 0.2398069576842853, -0.04788135162036506, 0.04752205982468178]
> Traceback (most recent call last):
>   File "example_1.py", line 23, in <module>
>     rob.movej((0.1, 0.1, 0.1, 0.1, 0.1, 0.1), 0.5, 0.1)
>   File "/usr/local/lib/python3.6/dist-packages/urx-0.11.0-py3.6.egg/urx/urrobot.py", line 330, in movej
>     self._wait_for_move(joints[:6], threshold=threshold, joints=True)
>   File "/usr/local/lib/python3.6/dist-packages/urx-0.11.0-py3.6.egg/urx/urrobot.py", line 270, in _wait_for_move
>     raise RobotException("Robot stopped")
> urx.urrobot.RobotException: Robot stopped
> 

thank you very much.

Abner-fu, 2024-06-13:
In URsim, you can use the urx library for simulation control. However, because the URsoftware used in URsim does not support setting up the network, the is_running() in the urx library is always judged to be False, so you need to modify the files in the urx (0.11.0) library.

1. First ensure that URsim can be pinged by the host computer, that is, URsim can communicate with the host computer.
2. There is an ursecmon.py file in the directory where urx was downloaded.  Change this line [https://github.com/SintefManufacturing/python-urx/blob/e277ef1a7fb057f9e642a21cd95ade2b4e5a3468/urx/ursecmon.py#L316](https://github.com/SintefManufacturing/python-urx/blob/e277ef1a7fb057f9e642a21cd95ade2b4e5a3468/urx/ursecmon.py#L316) to
   `and self._dict["RobotModeData"]["isRobotConnected"] is False \`
3. If you are using URsoftware version >=3.5, refer to [https://github.com/SintefManufacturing/python-urx/commit/e1f78ce203dcc8c4f1ed45ccf51e8a105f103912](https://github.com/SintefManufacturing/python-urx/commit/e1f78ce203dcc8c4f1ed45ccf51e8a105f103912). The author submitted the update, but the update is not reflected in the `pip install urx`. So you can add these three lines manually.
4. If you use movel or getl function, you may encounter AttributeError: 'PoseVector' object has no attribute 'tolist'. Please refer to [https://github.com/SintefManufacturing/python-urx/issues/117#issuecomment-1852557635](https://github.com/SintefManufacturing/python-urx/issues/117#issuecomment-1852557635)

samyk, 2026-01-21:
math3d v4 (and thus `movec` and other functions) supported here: https://github.com/samyk/python-urx/