SintefManufacturing/python-urx #103: Can I remotely call a .urp program written on teach pendant using urx?

srujan-aris, 2021-11-11:
@oroulet 
I have a program (program.urp) written on the ur10e teach pendant.
I want to call this program from a remote machine using the program name. Is it possible with this library, may be using the send_program() method?

robot-ranger, 2021-12-04:
Do you need to use urx specifically? Or just remote in general?

you can do this over the dashboard server; no reason you couldnt have python send the string :
[](https://www.universal-robots.com/articles/ur/dashboard-server-e-series-port-29999/)

srujan-aris, 2021-12-04:
> Do you need to use urx specifically? Or just remote in general?
> 
> you can do this over the dashboard server; no reason you couldnt have python send the string :
> [Dashboard Server](https://www.universal-robots.com/articles/ur/dashboard-server-e-series-port-29999/)

Hi. Thanks for your reply. I was looking for a solution to control remotely. Not particularly with urx. I eventually figured out about the dashboard server after I opened this issue. This can be closed now. Thanks again.

closed 2021-12-04 by srujan-aris