SintefManufacturing/python-urx #109: movej <wait=True> not working

danielghirst, 2022-06-17:
Versions:
Python 3.8.10
urx        0.11.0

Issue: 
movej does not wait when I set wait to wait = True.

Code snippet:

`def goto_loading_position(robot):
    pos = ((2*pi), (1.117011), (-1.797689), (-0.8726646), (0.5*pi), (2*pi)) #works with testing mag cal
    vel = 0.5
    acc = 0.5

    #Goto position
    try:
        robot.movej(pos, acc=acc, vel=vel, wait=True, relative=False, threshold=None)
    except Exception as e:
        print("Robot could not execute goto loading position (emergency stop for example), do something")
        print("\nException:", e, "\n")

> ./robot_loading_position_cmd.py
tried 11 times to find a packet in data, advertised packet size: -2, type: 3
Data length: 68
tried 11 times to find a packet in data, advertised packet size: -2, type: 3
Data length: 1092
tried 11 times to find a packet in data, advertised packet size: -2, type: 3
Data length: 1444
Robot could not execute goto loading position (emergency stop for example), do something

Exception: Robot stopped

It looks like that exception comes from urrobot.py??

`if not self.is_running():
                raise RobotException("Robot stopped")`

danielghirst, 2022-06-20:
Update: 

When I change `if self._dict["RobotModeData"]["robotMode"] == rmode\` to `if self._dict["RobotModeData"]["robotMode"] == 7 \`
I no longer have the issue.

yongpan0715, 2023-05-21:
Thank you, it is useful, successfully solved my problem