SintefManufacturing/python-urx #112: Problem of urx.urrobot.RobotException: Robot stopped

lainaddeys, 2023-03-01:
When I try to run a command ie robot.translate((0.5,0.5,0.5),0.1,0.1), the error always comes out "urx.urrobot.RobotException: Robot stopped" after the robot completes the movement. When I try to run a code like this:
def move():
     robot.translate((0.5,0.5,0.5),0.1,0.1)
     robot.movel((0.05,0.05,0.05,0,0,0),0.1,0.1)

It completes the first movement and proceeds with the error. I was wondering how do I fix this as my objective movements are much more complex. I installed the library by pip install urx on my Raspberry Pi. I hope someone can help me thank you.

l-okkkkk-k, 2023-03-02:
I have the same problem. if you solve it please tell me

canismarko, 2023-08-05:
I ran into this problem and realized it was solved on the the github version in commit e1f78ce203dcc8c4f1ed45ccf51e8a105f103912.

The version on pypi (0.11.0) doesn't parse version > 3.5, and my robot had 3.7 (or something like that). So it assumed you're using version 0.0 and then doesn't read the running mode of the robot correctly.

Does the problem go away if you do 

```
pip install git+https://github.com/SintefManufacturing/python-urx.git
```
?