SintefManufacturing/python-urx #114: how to get the result of calling `get_inverse_kin_has_solution`

blackhorsewu, 2023-06-29:
I want to check poses are reachable before moving the robot to it. URScript starting from version 3.15 provide a function `get_inverse_kin_has_solution`. I know how to send that as a program to the robot but do not know how to get the result if the pose is reachable or not. Any help would be very much appreciated.

ShanMallinathan, 2023-07-04:
I could not find this functionality with the urx package but I was able to run this using a socket connection with the robot manually. 
Cheers!

blackhorsewu, 2023-07-05:
> @blackhorsewu I could not find this functionality with the urx package but I was able to run this using a socket connection with the robot manually. Follow the below link for more reference https://www.zacobria.com/universal-robots-knowledge-base-tech-support-forum-hints-tips-cb2-cb3/index.php/universal-robots-script-client-server-example/ Cheers!

@ShanMallinathan Thank you very much for your reply. My problem is how can I read the result? It should be a Boolean, either True or False. How can this be sent back to my PC running python?

ShanMallinathan, 2023-07-05:
@blackhorsewu 
Store the boolean in a variable in the URScript and send it to the server (PC) using the socket_send_string command. receive it on your PC using recv(). I tried it in the simulator and it worked just fine.

blackhorsewu, 2023-07-06:
> @blackhorsewu Store the boolean in a variable in the URScript and send it to the server (PC) using the socket_send_string command. receive it on your PC using recv(). I tried it in the simulator and it worked just fine.

@ShanMallinathan
 Thank you very much and I will try it.