SintefManufacturing/python-urx #117: AttributeError: 'PoseVector' object has no attribute 'tolist'

YHt666, 2023-09-24:
The python version is 3.11. The problem is as follows:
C:\Users\YHT\.conda\envs\ur5py311\python.exe E:/UR/myproj1/test.py
tried 11 times to find a packet in data, advertised packet size: -2, type: 3
Data length: 44
tried 11 times to find a packet in data, advertised packet size: -2, type: 3
Data length: 68
tried 11 times to find a packet in data, advertised packet size: -2, type: 3
Data length: 1092
tried 11 times to find a packet in data, advertised packet size: -2, type: 3
Data length: 1418
Traceback (most recent call last):
  File "E:\UR\myproj1\test.py", line 10, in <module>
    pose = rob.getl()           # x,y,z,rx,ry,rz
           ^^^^^^^^^^
  File "C:\Users\YHT\.conda\envs\ur5py311\Lib\site-packages\urx\robot.py", line 211, in getl
    return t.pose_vector.tolist()
           ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'PoseVector' object has no attribute 'tolist'

TrinhNC, 2023-09-26:
got the same error. have you solved it?

MathijsNL, 2023-09-28:
I also got this error.

It seems related to the addition of the math3d library. The following function will also give an error:

```
rob.getl()
```

It seems in general there has been some degradation in working with tposes. I will try to dive into it somewhere this or next week.

MathijsNL, 2023-09-28:
In the meantime, downgrading to version 0.8.2 works fine if you change [this line](https://github.com/SintefManufacturing/python-urx/blob/0.8.2/urx/robot.py#L15) to False:

```
MATH3D = False
```

sddfsAv, 2023-10-18:
I modified the robot.py:

`t.pose_vector.tolist()` => `t.pose_vector.get_array().tolist()`

The error may be avoided

axi92, 2023-10-24:
When you change this line https://github.com/SintefManufacturing/python-urx/blob/0.11.0/urx/robot.py#L211
to:
`        return t.pose_vector.get_array().tolist()`
it works again.

jorge5a, 2023-11-03:
It didn't works with this change 
     return t.pose_vector.get_array().tolist()

I use the last version. Another idea?

YHt666, 2023-11-04:
I have chosen another package called [Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver). No problems so far. You can try it out. Good luck!

FLandolt, 2023-11-20:
> In the meantime, downgrading to version 0.8.2 works fine if you change [this line](https://github.com/SintefManufacturing/python-urx/blob/0.8.2/urx/robot.py#L15) to False:
> 
> ```
> MATH3D = False
> ```

HI this worked but have you found a better solution in the mean time ? :D

n4z3r, 2023-11-21:
Please fix!! I got it too

n4z3r, 2023-11-21:
> > In the meantime, downgrading to version 0.8.2 works fine if you change [this line](https://github.com/SintefManufacturing/python-urx/blob/0.8.2/urx/robot.py#L15) to False:
> > ```
> > MATH3D = False
> > ```
> 
> HI this worked but have you found a better solution in the mean time ? :D

This does not allow movej to work. It lets movel to work though. We need all functionality please.

se-hwan, 2023-12-12:
Also had this issue, but fixed it in `urx` version 0.11.0 with the following fix:

Change this line: https://github.com/SintefManufacturing/python-urx/blob/c5dba8594ca2fd09a29ca433171ad09956fb1482/urx/urrobot.py#L313
to this: `prog = self._format_move(command, tpose.array, acc, vel, prefix="p")`

Similarly, modify this line: https://github.com/SintefManufacturing/python-urx/blob/c5dba8594ca2fd09a29ca433171ad09956fb1482/urx/urrobot.py#L316C13-L316C64
to this: `self._wait_for_move(tpose.array[:6], threshold=threshold)`

`tpose` seems to originally have been some iterable data structure, but is now a `PoseVector` class from the `m3d` package. Calling the `array` method on this class returns an iterable list, which is what the `urx` methods seem to expect. 

It's likely you'll get a constant stream of terminal output warning about deprecation with the `dist_squared()` function. I commented out this warning in line 280 of `vector.py` in the `math3d` package (wherever it is in your virtual environment, system folder, etc.) to suppress this.

I tested that the `movel()` and `translate()` functions work, but it still causes an error with the `translate_tool()` function. Generally, I think a lot of code needs to be refactored to make everything work smoothly with the `m3d` libraries, but hope this helps.

zp2546265641, 2024-07-18:
> Also had this issue, but fixed it in `urx` version 0.11.0 with the following fix:
> 
> In addition to the fix here: [#117 (comment)](https://github.com/SintefManufacturing/python-urx/issues/117#issuecomment-1777285587), which addresses using the `getl()` method, this should address using the `movel()` and `translate()` methods for the UR5 arms.
> 
> Change this line:
> 
> https://github.com/SintefManufacturing/python-urx/blob/c5dba8594ca2fd09a29ca433171ad09956fb1482/urx/urrobot.py#L313
> 
> 
> to this: `prog = self._format_move(command, tpose.array, acc, vel, prefix="p")`
> Similarly, modify this line:
> 
> https://github.com/SintefManufacturing/python-urx/blob/c5dba8594ca2fd09a29ca433171ad09956fb1482/urx/urrobot.py#L316
> 
> 
> to this: `self._wait_for_move(tpose.array[:6], threshold=threshold)`
> `tpose` seems to originally have been some iterable data structure, but is now a `PoseVector` class from the `m3d` package. Calling the `array` method on this class returns an iterable list, which is what the `urx` methods seem to expect.
> 
> It's likely you'll get a constant stream of terminal output warning about deprecation with the `dist_squared()` function. I commented out this warning in line 280 of `vector.py` in the `math3d` package (wherever it is in your virtual environment, system folder, etc.) to suppress this.
> 
> I tested that the `movel()` and `translate()` functions work, but it still causes an error with the `translate_tool()` function. Generally, I think a lot of code needs to be refactored to make everything work smoothly with the `m3d` libraries, but hope this helps.
great work you do! happy!

samyk, 2026-01-21:
There are several more bugs like this, especially when using math3d v4. Looks like repo isn't maintained anymore so allPoseVector related bugs I ran into are fixed here: https://github.com/samyk/python-urx/