SintefManufacturing/python-urx #123: 'PoseVector' object is not iterable when using movel

n-conq, 2024-04-10:
When using movel, I always get a not iterable-error. Has anyone had a similar problem?

```
from urx import Robot
from time import sleep

rob = Robot("169.254.1.2")

#tool center point (tcp) in (x,y,z,rx,ry,rz)
rob.set_tcp((0, 0, 0.1, 0, 0, 0))
#payload in (kg, (center of gravity x, cog y, cog z))
rob.set_payload(0.85, (0, 0, 0.1))
sleep(0.5)
print("Current tool pose is: ",  rob.getl())

x, y, z = 0, 0, -0.1
rx, ry, rz = 0, 0, 0
a, v = 0.1, 0.1
rob.movel((x, y, z, rx, ry, rz), a, v, relative= True, wait=True)

sleep(0.5)

exit()
```

Current tool pose is:  [0.4542542097095572, -0.15457210675762453, -0.027859574520904748, 1.5646499092156947, -2.4645236374310726, -0.6841622895865939]
Traceback (most recent call last):
  File "c:\Users\Sophie\OneDrive\Desktop\Promotion\CryoTec\UR5_control", line 18, in <module>
    rob.movel((x, y, z, rx, ry, rz), a, v, relative= True, wait=True)
  File "C:\Users\Sophie\AppData\Local\Programs\Python\Python312\Lib\site-packages\urx\urrobot.py", line 284, in movel
    return self.movex("movel", tpose, acc=acc, vel=vel, wait=wait, relative=relative, threshold=threshold)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Sophie\AppData\Local\Programs\Python\Python312\Lib\site-packages\urx\robot.py", line 178, in movex
    return self.add_pose_base(t, acc, vel, wait=wait, command=command, threshold=threshold)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Sophie\AppData\Local\Programs\Python\Python312\Lib\site-packages\urx\robot.py", line 114, in add_pose_base
    return self.set_pose(trans * pose, acc, vel, wait=wait, command=command, threshold=threshold)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Sophie\AppData\Local\Programs\Python\Python312\Lib\site-packages\urx\robot.py", line 105, in set_pose
    pose = URRobot.movex(self, command, t.pose_vector, acc=acc, vel=vel, wait=wait, threshold=threshold)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Sophie\AppData\Local\Programs\Python\Python312\Lib\site-packages\urx\urrobot.py", line 313, in movex
    prog = self._format_move(command, tpose, acc, vel, prefix="p")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Sophie\AppData\Local\Programs\Python\Python312\Lib\site-packages\urx\urrobot.py", line 299, in _format_move        
    tpose = [round(i, self.max_float_length) for i in tpose]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'PoseVector' object is not iterable

flonzu, 2024-04-12:
I have the same error. My code was actually running without such an error six months ago on another computer. The code has not been changed since then.

1538933805, 2024-05-10:
maybe math3d is too new, try 3.3.x or 3.4.x？ I had the same problem befor I uninstall math3d==4.0.0 and then install math3d==3.3.0

1538933805, 2024-05-10:
Maybe math3d is too new, try 3.3.x or 3.4.x？ I had the same problem befor I uninstalled math3d==4.0.0 and then installed math3d==3.3.0

lemio, 2024-06-03:
I solved it by doing this, but some features are not available in 0.8.2 

In the meantime, downgrading to version 0.8.2 works fine if you change [this line](https://github.com/SintefManufacturing/python-urx/blob/0.8.2/urx/robot.py#L15) to False:

MATH3D = False

Adelina033, 2025-03-10:
Has anyone found a solution to this problem?

Jiabao-Zhao, 2025-06-24:
I believe these changed would help 

1. urx=0.11.0 version
2. in robot.py, below change needs to be made: def getl(self, wait=False, _log=True): """ return current transformation from tcp to current csys """ t = self.get_pose(wait, _log) return t.pose_vector.get_array().tolist()
3. in rurobot.py, below change is required: def movex(self, command, tpose, acc=0.01, vel=0.01, wait=True, relative=False, threshold=None): """ Send a move command to the robot. Since UR robots have several methods, this sends whatever is defined in 'command' string (e.g., "movel", "movep"). """
4. if hasattr(tpose, "array"): tpose = tpose.array
5. if relative: l = self.getl() tpose = [v + l[i] for i, v in enumerate(tpose)]
6. prog = self._format_move(command, tpose, acc, vel, prefix="p") self.send_program(prog)
7. if wait: self._wait_for_move(tpose[:6], threshold=threshold) return self.getl()

samyk, 2026-01-21:
There are several more bugs like this, especially when using math3d v4. Looks like repo isn't maintained anymore so all PoseVector related bugs I ran into are fixed here: https://github.com/samyk/python-urx/