SintefManufacturing/python-urx #124: Data packet timeout

fredster999, 2024-04-24:
I occasionally get timeout errors for urx supposedly not receiving a valid data packet within the timeout window and I'm not sure why it happens. Happens rather infrequently and doesnt seem to have any pattern for when it happens.

```
File "C:\Users\acryl\AppData\Local\pypoetry\Cache\virtualenvs\ur5-KN_aBXqf-py3.9\lib\site-packages\urx\urrobot.py", line 570, in movels
    return URRobot.movexs(
  File "C:\Users\acryl\AppData\Local\pypoetry\Cache\virtualenvs\ur5-KN_aBXqf-py3.9\lib\site-packages\urx\urrobot.py", line 661, in movexs
    self._wait_for_move(
  File "C:\Users\acryl\AppData\Local\pypoetry\Cache\virtualenvs\ur5-KN_aBXqf-py3.9\lib\site-packages\urx\urrobot.py", line 282, in _wait_for_move
    dist = self._get_dist(target, joints)
  File "C:\Users\acryl\AppData\Local\pypoetry\Cache\virtualenvs\ur5-KN_aBXqf-py3.9\lib\site-packages\urx\urrobot.py", line 307, in _get_dist
    return self._get_lin_dist(target)
  File "C:\Users\acryl\AppData\Local\pypoetry\Cache\virtualenvs\ur5-KN_aBXqf-py3.9\lib\site-packages\urx\urrobot.py", line 311, in _get_lin_dist
    pose = URRobot.getl(self, wait=True)
  File "C:\Users\acryl\AppData\Local\pypoetry\Cache\virtualenvs\ur5-KN_aBXqf-py3.9\lib\site-packages\urx\urrobot.py", line 515, in getl
    pose = self.secmon.get_cartesian_info(wait)
  File "C:\Users\acryl\AppData\Local\pypoetry\Cache\virtualenvs\ur5-KN_aBXqf-py3.9\lib\site-packages\urx\ursecmon.py", line 676, in get_cartesian_info
    self.wait()
  File "C:\Users\acryl\AppData\Local\pypoetry\Cache\virtualenvs\ur5-KN_aBXqf-py3.9\lib\site-packages\urx\ursecmon.py", line 668, in wait
    raise TimeoutException(
urx.ursecmon.TimeoutException: Did not receive a valid data packet from robot in 60
```

Timeout is 60 because I tried to see if raising it would help but it hasn't seemed to have done anything.

Any help would be appreciated.

Jsola052, 2024-04-29:
I had this same issue, I got around by using a loop that attempts to connect to the robot 5 times. Once connected it performs the rest of the code.

qchihiro, 2025-05-22:
I have also encountered the URX timeout error issue. Could you please let me know if there's a solution to this problem?